teemuatlut/TMCStepper #277: D1 Mini (ESP8266), TMC2208 - throwing exception

Mazvy, 2023-06-11:
Hey all,

I'm using the example "Simple" sketch and as soon as I use `driver.begin();` it always results in exceptions thrown.

Sorry if I'm missing something obvious, but can't figure out what's wrong?

**Sketch:**
```c++
#include <Arduino.h>
#include <TMCStepper.h>

// TMC
#define EN_PIN           D4 // Enable
#define DIR_PIN          D2 // Direction
#define STEP_PIN         D3 // Step
#define SW_RX            D7 // TMC2208/TMC2224 SoftwareSerial receive pin
#define SW_TX            D8 // TMC2208/TMC2224 SoftwareSerial transmit pin
#define R_SENSE 0.11f

TMC2208Stepper driver = TMC2208Stepper(SW_RX, SW_TX, R_SENSE); // Software serial

bool shaft = false;

void setup() {

	Serial.begin(9600);

	pinMode(EN_PIN, OUTPUT);
	pinMode(STEP_PIN, OUTPUT);
	pinMode(DIR_PIN, OUTPUT);
	digitalWrite(EN_PIN, LOW);

	//driver.beginSerial(115200);
	driver.begin();

	// driver.toff(5); 
	// driver.rms_current(600);
	// driver.microsteps(16);

	// driver.en_spreadCycle(false);
	// driver.pwm_autoscale(true);	
}

void loop() {	
	// Run 5000 steps and switch direction in software
	// for (uint16_t i = 5000; i>0; i--) {
	// 	digitalWrite(STEP_PIN, HIGH);
	// 	delayMicroseconds(160);
	// 	digitalWrite(STEP_PIN, LOW);
	// 	delayMicroseconds(160);
	// }
	// shaft = !shaft;
	// driver.shaft(shaft);
}
```

**Exception:**

```
Exception (28):
epc1=0x40203e83 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000010 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 0190
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  00000010 feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe 00000001  
3ffffeb0:  00000000 3ffef374 3ffef374 40203f29  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe 3fffff30 40106be0  
3ffffee0:  000000bd 000000bd 3ffe85dc 40100933  
3ffffef0:  000000c9 000000c9 3ffe85dc 40100933  
3fffff00:  3fffdad0 00000020 3ffef52c 3ffee598  
3fffff10:  3fffdad0 00000000 3ffee4cc 40204016  
3fffff20:  3ffef58c 00000000 3ffef58c 402055a0  
3fffff30:  3fffda05 3ffee610 0000000d 4020c030  
3fffff40:  3ffef40c 00000000 3ffef374 40205285  
3fffff50:  feefeffe 00000000 3ffef374 40201326  
3fffff60:  00800005 78410100 3ffee4cc 40201132  
3fffff70:  00000040 00000000 3ffee4cc 40205190  
3fffff80:  00002580 0000001c 3ffee4cc 40201157  
3fffff90:  3fffdad0 00000000 3ffee584 40201065  
3fffffa0:  feefeffe feefeffe feefeffe 40201df8  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100e99  
<<<stack<<<
```

**Decoded exception:**

```log
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 16 results
0x40203e83: circular_queue ::for_each(Delegate  const&) at ?? line ?
0x40203f29: SoftwareSerial::rxBits() at ?? line ?
0x40106be0: SoftwareSerial::write(unsigned char const*, unsigned int, SoftwareSerialParity) at ?? line ?
0x40100933: umm_free_core at umm_malloc.cpp line ?
0x40100933: umm_free_core at umm_malloc.cpp line ?
0x40204016: SoftwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x402055a0: SoftwareSerial::write(unsigned char) at ?? line ?
0x4020c030: operator delete(void*, unsigned int) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_ops.cc line 34
0x40205285: TMC2208Stepper::serial_write(unsigned char) at ?? line ?
0x40201326: TMC2208Stepper::write(unsigned char, unsigned int) at ?? line ?
0x40201132: TMC2208Stepper::beginSerial(unsigned int) at ?? line ?
0x40205190: TMC2208Stepper::pdn_disable(bool) at ?? line ?
0x40201157: TMC2208Stepper::begin() at ?? line ?
0x40201065: setup at ?? line ?
0x40201df8: loop_wrapper() at core_esp8266_main.cpp line ?
0x40100e99: cont_wrapper at ?? line ?
```

daniel-frenkel, 2023-06-11:
What microcontroller are you using?

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 0190
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  00000010 feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe 00000001  
3ffffeb0:  00000000 3ffef374 3ffef374 40203f29  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe 3fffff30 40106be0  
3ffffee0:  000000bd 000000bd 3ffe85dc 40100933  
3ffffef0:  000000c9 000000c9 3ffe85dc 40100933  
3fffff00:  3fffdad0 00000020 3ffef52c 3ffee598  
3fffff10:  3fffdad0 00000000 3ffee4cc 40204016  
3fffff20:  3ffef58c 00000000 3ffef58c 402055a0  
3fffff30:  3fffda05 3ffee610 0000000d 4020c030  
3fffff40:  3ffef40c 00000000 3ffef374 40205285  
3fffff50:  feefeffe 00000000 3ffef374 40201326  
3fffff60:  00800005 78410100 3ffee4cc 40201132  
3fffff70:  00000040 00000000 3ffee4cc 40205190  
3fffff80:  00002580 0000001c 3ffee4cc 40201157  
3fffff90:  3fffdad0 00000000 3ffee584 40201065  
3fffffa0:  feefeffe feefeffe feefeffe 40201df8  
3fffffb0:  feefeffe feefeffe 3ffe85d8 40100e99  
<<<stack<<<

Decoded exception:
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 16 results
0x40203e83: circular_queue ::for_each(Delegate  const&) at ?? line ?
0x40203f29: SoftwareSerial::rxBits() at ?? line ?
0x40106be0: SoftwareSerial::write(unsigned char const*, unsigned int, SoftwareSerialParity) at ?? line ?
0x40100933: umm_free_core at umm_malloc.cpp line ?
0x40100933: umm_free_core at umm_malloc.cpp line ?
0x40204016: SoftwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x402055a0: SoftwareSerial::write(unsigned char) at ?? line ?
0x4020c030: operator delete(void*, unsigned int) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_ops.cc line 34
0x40205285: TMC2208Stepper::serial_write(unsigned char) at ?? line ?
0x40201326: TMC2208Stepper::write(unsigned char, unsigned int) at ?? line ?
0x40201132: TMC2208Stepper::beginSerial(unsigned int) at ?? line ?
0x40205190: TMC2208Stepper::pdn_disable(bool) at ?? line ?
0x40201157: TMC2208Stepper::begin() at ?? line ?
0x40201065: setup at ?? line ?
0x40201df8: loop_wrapper() at core_esp8266_main.cpp line ?
0x40100e99: cont_wrapper at ?? line ?

@daniel-frenkel tagging you :)

Mazvy, 2023-06-11:
@daniel-frenkel D1 Mini (ESP8266)

daniel-frenkel, 2023-06-11:
That appears to be an issue with the 8266 and Serial and not this library. You might need to include the software serial library. Unfortunately this is out of my scope but the 8266 forum will probably help

Mazvy, 2023-06-12:
Thank you! I tried using a software library, but `driver.test_connection();` always returns `2`.

Perhaps I'm setting the sketch up wrong?

```c++
#include <Arduino.h>
#include <TMCStepper.h>
#include <SoftwareSerial.h>

// TMC
#define EN_PIN           D4 // Enable
#define DIR_PIN          D2 // Direction
#define STEP_PIN         D3 // Step
#define SW_RX            D7 // TMC2208/TMC2224 SoftwareSerial receive pin
#define SW_TX            D8 // TMC2208/TMC2224 SoftwareSerial transmit pin
#define R_SENSE 0.11f

SoftwareSerial TMC(SW_RX, SW_TX);
TMC2208Stepper driver = TMC2208Stepper(&TMC, R_SENSE); // Software serial

void setup() {
	Serial.begin(9600);	
	driver.begin();
	driver.toff(5); 
}

void loop() {	
    delay(1000);
    uint8_t result = driver.test_connection();
    Serial.println(result);	
}
```

daniel-frenkel, 2023-06-17:
@Mazvy 

Please post your schematic

chrishdx, 2023-07-07:
I had the same problem with an ESP8266; I couldn't get it to work. However, with an ESP32, I used GPIO 16 and 17, which correspond to SerialPort 2. However, I had to put a resistor between TX and RX. 1K Ohm didn't work; I tested 470 ±5% and 560 ±5%, both worked. For the TMC, I tested two variants, one with RX and TX, where I had to use the RX pin. In the other variant, both PD pins were directly connected, so it didn't matter.

Mazvy, 2023-07-07:
> I had the same problem with an ESP8266; I couldn't get it to work. However, with an ESP32, I used GPIO 16 and 17, which correspond to SerialPort 2. However, I had to put a resistor between TX and [...]

I also switched to ESP32 and with some tweaking (namely this https://github.com/teemuatlut/TMCStepper/issues/192#issuecomment-831432951) it's been working fine. I've yet to implement stall guard, but I'm controlling the motor without any issues thus far solely via UART.

chrishdx, 2023-07-08:
Today, I also managed to do it with an ESP8266. For this, I swapped the serial port to D7 (rx) and D8 (tx) and created a SoftwareSerial on the old serial port with 9600 baud. Whenever I used a SoftwareSerial of 115200, the ESP would always crash, but 9600 is enough for the connection in my case. I have directly connected the EN pin to GND. Attached is my code:  """ 

#include <TMCStepper.h>

#ifdef ESP32
#define TMC_Serial Serial2
#define User_Serial Serial
#else
#include <SoftwareSerial.h>
#define TMC_Serial Serial
SoftwareSerial User_Serial (3,1);// SoftwareUser_Serial port for other boards (RX, TX)
//#define EN_PIN 2 // Enable
#endif

#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2
#define R_SENSE 0.11f // Match to your driver
int stallVal = 0;
bool startup = true; // set false after homing

TMC2209Stepper driver(&TMC_Serial, R_SENSE, DRIVER_ADDRESS);
 
bool shaftVal = false;
bool stalled_X = false;

void setup() {

//pinMode(EN_PIN, OUTPUT);
//digitalWrite(EN_PIN, LOW);

TMC_Serial.begin(115200);
TMC_Serial.swap();
delay(500);
User_Serial.begin(9600);
User_Serial.println("TMC2209 Stallguard Test");
driver.begin(); // SPI: Init CS pins and possible SW SPI pins
driver.toff(4); // Enables driver in software, changed from 5
driver.blank_time(24);
driver.rms_current(700); // Set motor RMS current
driver.microsteps(2); // Set microsteps to 1/16th
driver.pwm_autoscale(true); // Needed for stealthChop

driver.semin(5);
driver.semax(2);
driver.sedn(0b01);
driver.TPOWERDOWN(1);
driver.isEnabled();
driver.shaft(shaftVal);
driver.TCOOLTHRS(0xFFFFF); // 20bit max
User_Serial.println("Driver Enabled");
User_Serial.println(driver.version());
User_Serial.println(" ");

}

void zu(){
  driver.VACTUAL(500);
  driver.shaft(false);
  delayMicroseconds(100);
  stalled_X = false;
  while (!stalled_X)
  {
    int wert = driver.SG_RESULT();
    User_Serial.println(wert);

    if (wert < 170 &&  wert >20){
        stalled_X = true;
        driver.VACTUAL(0);
    }
  }
  
}

void auf(){
  driver.VACTUAL(300);
  driver.shaft(true);
  delayMicroseconds(100);
  stalled_X = false;
  while (!stalled_X)
  {
    int wert = driver.SG_RESULT();
    User_Serial.println(wert);

    if (wert < 170 &&  wert > 142){
        stalled_X = true;
        driver.VACTUAL(0);

    }
  }
  
}

void loop() {
if (startup == true){
startup = false;
auf();
}

}

"""