teemuatlut/TMCStepper #279: Is it possible to use all the features of TMC2209 straight from an Arduino Uno or an ESP32 (classic)? Or is an extra motor controller (such as the CNC shield)?

01GOD, 2023-06-23:
Is it possible to use all the features of TMC2209 straight from an Arduino Uno or an ESP32 (classic)? Or is an extra motor controller (such as the CNC shield)?

daniel-frenkel, 2023-06-26:
@01GOD Yes it is, no extra hardware is required. The ESP32 is far superior for running a stepper motor because it has[ built in hardware ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html) to do it

yagnik, 2023-09-01:
@01GOD esp32 gpio pins are rated 40 mA ( happy path at 20 mA ) with 3.3V. 
Most stepper motors are rated higher ( > 1A for 3d printers as an example ) so some additional circuitry would be needed in order to drive enough current in the coil.
For this reason, generally a driver is needed which can pass higher power to the coils.
The esp32 can still be used to provide a pwm signal (by manually building the signal or using the ledc or motorpwm channels ) along with 2209 to provide power.

Not directly needed but additional:
- tmc 2209  does not provide ramping up/down ( acceleration ) so that has to be handled manually or a different driver has to be used like tmc 5160.
Hope this helps