teemuatlut/TMCStepper #283: class TMC5160Stepper' has no member named 'SGTHRS

MarkEvens, 2023-09-27:
What is replacement for this in TMC5160?

WillemBrx, 2024-01-23:
Same problem. Can't do anything with stallguard

cattledogGH, 2024-01-27:
`void sgt (int8_t B)`
`void sfilt (bool B)`
`uint16_t sg_result ()`

These Stallguard functions are all Public Member Functions inherited from TMC2130Stepper.
It is indeed possible to use Stallguard with the TMC5160.

WillemBrx, 2024-01-27:
Thank you. Could you give me an example on how to use that? maybe I can send you the code to take a look at? I would be very grateful!

cattledogGH, 2024-01-27:
Typically, I have first monitored the running stepper and determined the sg_result value when running and when stalled. Then in the application I have read the sg_result and compared it to a value and if it's less than that value I call it a stall detected.

If you want to post some code for review, do it here, and make a minimal, reproducible, example which demonstrates your issue.

WillemBrx, 2024-01-28:
Thank you!
I am using Adafruit ESP32 Feather V2

The code:

`
#include <TMCStepper.h>
#include <AccelStepper.h>
#include <SPI.h>

#define EN_PIN 33            // Enable
#define CS_PIN 15            // Chip select
#define STEP_PIN 32        // Step
#define DIR_PIN 14          // Direction
#define SW_MOSI 19       // Software Master Out Slave In (MOSI)
#define SW_MISO 21       // Software Master In Slave Out (MISO)
#define SW_SCK 5            // Software Slave Clock (SCK)
#define R_SENSE 0.075f   // stepper driver TMC5160 specific
#define STALL_PIN 27      // Connected to DIAG_0 
#define STALL_VALUE 15
TMC5160Stepper driver = TMC5160Stepper(CS_PIN, R_SENSE, SW_MOSI, SW_MISO, SW_SCK);
AccelStepper stepper = AccelStepper(stepper.DRIVER, STEP_PIN, DIR_PIN);

constexpr uint32_t stepper_motor_standard_steps_per_rotation = 200;
int micro_step = 32;
int MaxSpeed = 5000;
int Acceleration = 1000;

unsigned long lastMonitorTime = 0; // Variable to store the last time the monitoring was performed
unsigned long monitorInterval = 1000; // Interval for monitoring in milliseconds

void setup() {
  SPI.begin();
  Serial.begin(115200);
  
  pinMode(CS_PIN, OUTPUT);
  digitalWrite(CS_PIN, HIGH);
  pinMode(STALL_PIN, INPUT_PULLUP); //Adafruit ESP32 feather V2 internal pull up
  pinMode(EN_PIN, OUTPUT);
 digitalWrite(EN_PIN, LOW); 	                // Enable driver in hardware
pinMode(STEP_PIN, OUTPUT);

  Serial.println();
  Serial.print("DRV_STATUS=0x");
  Serial.println(driver.DRV_STATUS(), HEX);
  Serial.println(driver.IOIN());

  driver.begin();
  driver.rms_current(700);
  driver.en_pwm_mode(false);
  driver.pwm_autoscale(false);
  driver.microsteps(micro_step);

  stepper.setMaxSpeed(MaxSpeed);
  stepper.setAcceleration(Acceleration);
  stepper.setEnablePin(EN_PIN);
  stepper.setPinsInverted(false, false, true);
  stepper.enableOutputs();
  
  // stepper specs
  Serial.print("Default stepper motor steps per rotation: ");
  Serial.println(stepper_motor_standard_steps_per_rotation);

  float steps_per_rotation = driver.microsteps() * stepper_motor_standard_steps_per_rotation;
  Serial.print("Steps per Rotation: ");
  Serial.println(steps_per_rotation);

  Serial.print("micro step: ");
  Serial.println(micro_step);

  Serial.print("Max Speed (steps per second): ");
  Serial.println(stepper.maxSpeed());

  Serial.print("Acceleration (steps per second): ");
  Serial.println(Acceleration);

  float speed_rpm = (stepper.maxSpeed() / (steps_per_rotation / micro_step)) * 60.0;
  Serial.print("Max Speed (RPM): ");
  Serial.println(speed_rpm);

  
}

void loop() {
  if (stepper.distanceToGo() == 0) {
    stepper.disableOutputs();
    delay(1000);

    // Check the current direction and move accordingly
    if (stepper.currentPosition() == 0) {
      stepper.move(51200);
    } else {
      stepper.move(-51200);
    }

    stepper.enableOutputs();
  }

  stepper.run();
 // Check if it's time to perform monitoring
  unsigned long currentMillis = millis();
  if (currentMillis - lastMonitorTime >= monitorInterval) {
    // Update the last monitoring time
    lastMonitorTime = currentMillis;

    // Perform monitoring
    performMonitoring();
  }
}

void performMonitoring() {
  

  Serial.println();
  Serial.print(".cs_actual: ");
  Serial.println(driver.cs_actual());

  Serial.print("sg_result: ");
  Serial.println(driver.sg_result(), DEC);

  Serial.print("cs2rms(driver.cs_actual): ");
  Serial.println(driver.cs2rms(driver.cs_actual()), DEC);

  Serial.print("sg_result < STALL_VALUE: ");
  Serial.println(driver.sg_result() < STALL_VALUE, DEC);

}

`

WillemBrx, 2024-01-28:
[image]

When i hold the motor with my hand
[image]

WillemBrx, 2024-01-28:
The strange thing is that i do get a SG_result but only when I restart the esp32
[image]