teemuatlut/TMCStepper #304: Updated Resource to Help Those Struggling with the TMC2209

daniel-frenkel, 2024-09-10:
I've been working with the TMC2209 and other Trinamic drivers for over 6 years, and have written a guide to help get everyone started as quickly as possible, because these drivers and library are very complex. 

I've recently updated my TMC2209 guide to help those who are struggling with this library so I'm leaving this comment.

The guide is the Second Edition, which I originally wrote a few years ago and have recently updated. In the guide I cover the following:

1. Stepper motors 101: How they work
2. TMC2209 schematic and breadboard diagram
3. Understanding stepper motor power requirements
4. How to program the TMC2209 in Arduino with working examples
5. How to easily set up StallGuard
6. ESP32 Dual-core and single-core operations with an Arduino example
7. Finite state machine Arduino example
8. And much more.

If there's any confusion, just send me an email (located in the guide) I'll update the guide to help you guys learn more. Thanks!

[Link to Guide](https://amzn.to/3MCEPBI)

bluthen, 2024-10-13:
I was curious and the price was cheap. Been trying to get the TMC5106 and thought I could still learn from the TMC2209.

I've never tried publishing an ebook, however this is not readable on my kindle, or it seems like kindle web and insists I download a kindle app. Anyway to get a PDF or have it support more devices?

daniel-frenkel, 2024-10-13:
You have to read it on Kindle web because it’s a huge pain to format Arduino code for the Kindle reader. Regardless, I’ll send you a PDF copy if you email me at the email address that’s in the book.I don’t have any examples of using the motion controller like the TMC5160 but I’ve used it for many years and may add more info in the book about it. No one’s asked for it yet so I haven’t taken the time to write it all down. But like you said, they’re all similar devices and once you learn to use one you’ll learn to use them all.

bluthen, 2024-10-14:
Kindle for the web does not work for me either. It just says "We are sorry. Kindle can’t open this book.". Anyway don't mean for this to be a support thread. I'll delete my review on amazon. Good luck!

[image]

daniel-frenkel, 2024-10-14:
Well that’s really annoying, thanks for letting me know. I just assumed it’d work on the web reader but never tested it. I’ll see if I can change it

shot_frenkel.png (view on web)